michael-ring/pico-fpcexamples #1: serial communications

davidbannon, 2021-09-18:
Michael, like your work here.  
I wonder if it could be extended to use the serial interface ?  eg, the ADC example here seems to write the data it collects to the uart, that requires us to be working from a Raspberry Pi or via a debugging PicoProbe. The official Raspberry examples all seem to be able to switch between serial and uart - the serial approach being a lot more convenient for simple communication between a Pico and a desktop or laptop (running linux).

Any thoughts ?

Davo

michael-ring, 2021-09-19:
Using serial via USB adds more complexity to (example-) code, to make it work you need to constantly poll USB in your code otherwise the serial device will disappear after a short while. 

And for a pure embedded approach without any RTOS in the background it causes more issues than benefit because a simple sleep command that you do for any reason will already kill the communication when it is long enough.

I will do a FreeRTOS Version of pico with native Freepascal support without the need to link to so much C-Code libraries, there I will support Serial via USB as the whole polling can be done in an extra thread and does not disturb your coding flow.
But this will take a while, I am currently working on other topics.

davidbannon, 2021-09-20:
Thanks Michael. I just needed an A2D connected to my laptop, the RP2040 has better precision but I used an Arduino because I could communicate easily with it. So, my immediate problem has gone away but that connectivity issue with the RP2040 does still exist. Pity ...

Mind you, my quick experiments with the Pico and MicroPython (yes, I admit it, Python) were not that good anyway, was seeing many hundreds of counts zero error with a clip lead between input and ground. Even allowing for the (silly) scaling to 16bits, still a huge error. 

And, further, I see no real problem linking fpc to the Pico C libraries, we link to C libraries one way or another every time we run a FPC/Lazarus app, yours was a great idea IMHO.

Davo

closed 2022-08-16 by michael-ring