michael-ring/pico-fpcexamples #5: spi_st7735 line 27

cobaltBlueCloud, 2022-08-15:
Hi, 

sorry when I'm wrong.

First we declare 3 outputs:
  gpio_set_function(TPicoPin.SPI_CS,  TGPIOFunction.GPIO_FUNC_SPI);
  gpio_set_function(TPicoPin.SPI_SCK, TGPIOFunction.GPIO_FUNC_SPI);
  gpio_set_function(TPicoPin.SPI_TX,  TGPIOFunction.GPIO_FUNC_SPI);

These are 
CS   = gpio17 pin22
SCK = gpio18 pin24
TX   = gpio19 pin25

LINE 27
st7735.Initialize(spi,TPicoPin.GP16,TPicoPin.GP14,st7735.ScreenSize128x128x16);

What are GP16 and GP14 ?

Kind regards. Marcel

michael-ring, 2022-08-16:
Please check documentation in sources:

(*
      Initializes the display
    param
      SPI     The SPI Interface to use
      aPinDC  Pin used for switching between Communication between Data and Command Mode
      aPinRST Pin used to reset the display, not needed by all displays, pass TNativePin.None when not needed
      aPhysicalScreenInfo Information about Width/Height and Bitdepth of the connected screen
    note
      The SPI interface needs to be pre-initialized to required Parameters
      The extra Pins do not need to be initialized
    *)
    constructor Initialize(var SPI : TSpi_Registers;const aPinDC : TPinIdentifier;const aPinRST : TPinIdentifier;aPhysicalScreenInfo : TPhysicalScreenInfo);

closed 2022-08-16 by michael-ring

cobaltBlueCloud, 2023-11-02:
Hi,

[image: IMG_20231102_114243.jpg]
[image: pcb 03.jpg]
PCB is my own DIY.

Next ting is to dive into the CustomDisplay-unit!
Also got uart Tx working. Rx, I have to do next.

Kind regards. Marcel

cobaltBlueCloud, 2023-12-19:
Hi, Michael!

just ran into this issue:

line 38 in
https://github.com/cobaltBlueCloud/pico-fpcexamples/blob/main/uart_irq_demo/uart_irq.lpr

shows: received_string+=caractere;

should this be received_string:= received_string + caractere ?

Kind regards. Marcel