michael-ring/pico-fpcexamples #7: USB output

stalecu, 2022-11-25:
Hello! I'm trying to make a project where I am reading from and writing to the Pico (a REPL sort of thingie, doesn't matter the specifics). When I looked into the example, I could only find UART examples, but I need USB serial communication. The Raspberry Pi Pico SDK examples have a hello_usb example which is what I need, but they use a function called `stdio_init_all()`, but I'm not aware of an equivalent for this project. Neither `Writeln`, nor `Readln` works, as far as I can tell, and when I do use these functions along with the LED blinking, it blocks the LED, so I'm not sure of the interaction there. Could you at least point me in the right direction or add an example in the repo?

Thank you for your attention

michael-ring, 2022-11-25:
Unfortunately implementing USB is not a straightforward task because it requires background processes that keep the USB connection alive and also a number of extra libraries (tinyusb) needs to be brought in and the dependencies for those extra libraries will to be brought in, too.
I will look into this but chances are slim that this will be possible to do with reasonable effort.
I will close the issue for now and will reopen it when I had some success.

Sorry that there is no easy answer to your request.

closed 2022-11-25 by michael-ring

stalecu, 2023-02-07:
Sorry to reopen this, I could try thinking about this. What would one need to change or add for this functionality to work? Cuz I believe you could make tinyusb an optional dep (although I am pretty sure you have it by default when you pull the submodules in the Pico SDK). Also, why do you require background processes? Minicom handles it just fine, unless you mean on the Pico side (but I doubt that). Could you clarify, please?