michael-ring/pico-fpcexamples #9: Feeding back to official FPC

davidbannon, 2023-01-30:
Michael, I still love this work by the way !

However, we have an issue. Since December, Lazarus Main will no longer build with FPC320 and your compiler here is mostly FPC320. Some symbols were moved around apparently. Not unusual.

So, what really needs to happen is the important patches you have applied here need to find their way back into Lazarus Main, that would seem a lot easier than trying to bring this forked FPC up to FPC322 IMHO.

Is there anything that I can do to help with his ?

Davo

michael-ring, 2023-01-31:
Hi Davo, 

thanks for letting me know!

Usually it is not that much work to upgrade to a more recent version of fpc, I will give it a try this week.
I have discussed with the core team to bring my patches into fpc, but the amount of additional code for the device specific units is huge so it never happened.

Michael

davidbannon, 2023-01-31:
Thanks Michael, that would be a step in the right direction. 

But long term it does not seem like a great solution. It must be a fair bit of work for you to repeatedly retro fit your stuff like this.

Is there that much difference between the Pico arm and the other ones ? With FPC supporting so many platforms, I assumed it be relatively easy to add a platform once the details were all sorted out (as you have done already). 

Anyway, the change that has broken Lazarus exists only in main at present so, its easy to choose to use (eg Lazarus Fixes and Don says he will lock in an earlier Lazarus in fpcupdelux (for Pico) with the next release so, its workable as it is right now.

Thanks for all of this Michael, its good stuff !

Davo

michael-ring, 2024-03-03:
Christo Crause merged my changes in fpc

davidbannon, 2024-03-04:
Wow, Michael, that great news.
Thanks

I'll send you a unit and demo to get the LCD1602 display working shortly, just want to tidy it up a bit.

I have been playing with pico-W, with emphasis on "W" recently but not looking that good. Like 170 obj files and I still cannot get _sbrk to link !   Sigh....

Davo

michael-ring, 2024-03-04:
I just upgraded locally to latest 1.5.1 version of the Pico SDK, wanted to also give Pico-W a try but what you write does not sound that promising....