michael-ring/pico-fpcexamples #13: FPC main fails (for me) to build pico cross compiler

davidbannon, 2024-03-04:
Michael, I just tried building the **pico cross compiler** (linux, x86-64), on a FPC main source, downloaded today.

I think its more than possible I have my options wrong but I get a FPC Internal Error  200312126

```
/home/dbannon/bin/FPC/SRC/fpc-main/fpc/compiler/ppcrossarm -Ur -Parm -XParm-none-eabi- -Ur -Xs -O2 -n -Fi../inc -Fi../arm -Fi../unix -Fiarm -FD/usr/bin -FE. -FU/home/dbannon/bin/FPC/SRC/fpc-main/fpc/rtl/units/arm-linux -Fl/usr/lib/gcc/arm-none-eabi/9.2.1 -dFPC_ARMHF -CpARMV6M -CaEABIHF -darm -dRELEASE -CpARMV6M -CaEABIHF  -Us -Sg system.pp
system.inc(482,5) Fatal: Internal error 200312126
Fatal: Compilation aborted
make[5]: *** [Makefile:2873: system.ppu] Error 1
```

I am using these commands to build the cross compiler, the above error happens with the first one -
```
make clean all FPMAKEOPT="-T 4" CPU_TARGET=arm OPT="-dFPC_ARMHF" CROSSOPT="-CpARMV6M -CaEABIHF" CROSSBINDIR=/usr/bin BINUTILSPREFIX=arm-none-eabi-
```

Then, I would have run this one next -
```
make crossinstall OS_TARGET=linux CPU_TARGET=arm OPT="-dFPC_ARMHF" CROSSBINDIR=/usr/bin  BINUTILSPREFIX=arm-none-eabi-  CROSSOPT="-CpARMV6M -CaEABIHF" INSTALL_PREFIX=/home/dbannon/bin/FPC/fpc-3.3.1
```
Can you please comment on my compile lines ?  

Davo

michael-ring, 2024-03-04:
Can you try fpcupdeluxe? Have not built fpc+crosscompilers manually for ages.
What I can tell you is that using HF /Hardware Floating Point) togerther with ARMV6M is plain wrong as the poor little cortex-M0 does not have Hardware Floating point

closed 2024-03-04 by michael-ring

davidbannon, 2024-03-04:
Hmm, thats interesting !  Alf seems to use -dFPC_ARMHF.

Yes, the compiler in fpcupdeluxe seems to be OK but its a couple of years old, still pointing to your repo. And the Lazarus with it is terrible !  I have been using Lazarus 3.2 with your old compiler and all OK but thought I should test with something newer before submitting a unit to you.

But fpcupdeluxe has a clunky Lazarus and I thought if I built from scratch I could do some debugging for Alf and get fpcupdeluxe up to date.

Personally, I'm not a big fan of the black box approach, when something does not work as it should, you are left wondering whats in that black box. But Alf is very responsive so I might see what he thinks....

Davo

michael-ring, 2024-03-04:
fpcupdeluxe uses trunk for a while, my changes were merged to fpc's trunk. A recent version of fpcupdeluxe should reflect that change

davidbannon, 2024-03-04:
Not the latest fpcupdeluxe but three days ago ....

[image]