michael-ring/pico-fpcexamples #23: GPIO IRQ callback

jpmandon, 2024-06-09:
Hi Michael

in sdk1.5.1 branch

I tested the gpio_set_irq_enabled_with_callback function and I encountered a problem:

- First added irq_handler_chain.S.obj in the units-sdk1.5.5 folder ( it could be a future commit )
- now I have a SIGTRAP error when the gpio_set_irq_enabled_with_callback function is executed.
this is my code:
                // set IRQ enabled on signalPin and falling edge
                    gpio_set_irq_enabled_with_callback(
                        signalPin,
                        uint32(TGPIO_IRQ_level.GPIO_IRQ_EDGE_FALL),
                        true,
                        @signalHandlerTreatment
                    ); 

same problem when I try to use the gpio_set_irq_callback function. It seems that the problem is with the pointer of the callback.

any idea ?
Many thanks by advance.
Jean-Pierre

michael-ring, 2024-06-09:
Did you define your callback as interrupt?

procedure signalHandlerTreatment; interrupt; public name 'signalHandlerTreatment';

or similar

jpmandon, 2024-06-09:
I didn't declared the callback as interrupt but when I try to declare it as you suggested:
var
    signalHandler : TGPIO_Irq_callback ;
procedure signalHandlerTreatment(gpio:TPinIdentifier;events:longword);Interrupt;public name 'signalHandlerTreatment';
begin
end;

begin
signalHandler:=@signalHandlerTreatment;

I have an incompatible types message of the compiler.

In the hello_gpio_irq of the sdk the callback function is not declared as interrupt. So I didn't debug the C code but the only place where I find an assert is in the irq_add_shared_handler function. The documentation say:
This method will assert if there is an exclusive interrupt handler set for this irq number on this core, or if the (total across all IRQs on both cores) maximum (configurable via PICO_MAX_SHARED_IRQ_HANDLERS) number of shared handlers would be exceeded.
PICO_MAX_SHARED_IRQ_HANDLERS is set to 4 so it's not the problem.
I don't know if the IO_IRQ_BANK0_Handler is set as exclusive in the low level code of fpc and I'm not able to verify this, sorry.

I am going to write the same code with the sdk in C to see if it works !!